ANYCUBIC-3D/Vyper #1: configuration of TMC2209 drivers missing

FotoFieber, 2021-07-07:
According to the info on the anycubic website, the Vyper has TMC2209. The configuration in this github doesn't reflect that:
`#define X_DRIVER_TYPE  A4988`

KAPTOHKA, 2021-08-30:
[image]
Остаётся открытым вопрос о перемаркировке чипов )))

FotoFieber, 2021-08-31:
[image]

arnej, 2021-10-10:
Are there any news on this?

I replaced the noisy fans and now this is the only thing left that's loud on my machine.

I looked up what's going on with those steppers as they were marketed as quiet and found out that it should be possible to adjust and see how the drivers are configured with the M569 gcode command.
I tried this command, but the printer replies with "Unknown command", so I'm stuck there (I'm running stock firmware v2.3.5).

Is there anything I can do on my own to fix this or am I limited by hardware in this case?

SirHenry2, 2021-12-12:
how can something like this even happen? And why is no one talking about ?

wnsander232, 2022-01-01:
Take a look at StudioZombie 3D firmware.  He has made great progress on addressing some of these issues.  I'm running his ZombieVyper 2.5 baudfix (Dec 2, 2021) on one of my machines and have been very happy.  Here is his [LinkTree](https://linktr.ee/Zombiej79)

Its also worth noting he is very responsive to questions and assistance.  Good guy.