ANYCUBIC-3D/Vyper #2: Board file missing

phiradd, 2021-07-13:
Hi,

I was not able to compile the code neither with platformio nor with arduino ide because the board description is missing: ac_tri_f103re.

Please provide some information on how to successfully compile the code and or the respective json board file.

Thank you very much
Philipp

[image]

ANYCUBIC-3D, 2021-07-13:
> Hi,
> 
> I was not able to compile the code neither with platformio nor with arduino ide because the board description is missing: ac_tri_f103re.
> 
> Please provide some information on how to successfully compile the code and or the respective json board file.
> 
> Thank you very much
> Philipp
> 
> [image]

please refer to this building steps:
https://github.com/ANYCUBIC-3D/Vyper/wiki

FotoFieber, 2021-07-16:
Same problem here. Building steps on wiki didn't help.

chinamummy, 2022-01-13:
>
[image]
board_name AC_TRI_ F1
chip_name  STM32F103RET6

chaoshikong, 2022-04-11:
> > 
> 
> [image] board_name AC_TRI_ F1 chip_name STM32F103RET6

为什么我编译还是错误一大堆呢？

Mangers95, 2023-04-23:
Can someone explain what to do .. i dont get it to compilie :(

stopanko, 2023-11-11:
> Removed Platforms

What does it meant "Removed Platforms"