ANYCUBIC-3D/Vyper #3: Tool for erasing eeprom is missed

Semmi121, 2021-07-21:
The wiki talks about a tool to expand the bin file to erase the EEPROM. Unfortunately, there is no link to this tool.

mvoss96, 2021-08-03:
i was also searching for it but found it by downloading the wiki.
i hosted it on gdrive so you can just download it from here:https://drive.google.com/file/d/12BDNBJQoU1N1GMqRq2yjJENaakRY0C_V/view?usp=sharing

Semmi121, 2021-08-03:
Thanks a lot.

hajduzsolt, 2021-11-06:
How do you download the wiki?

> i was also searching for it but found it by downloading the wiki. i hosted it on gdrive so you can just download it from here:https://drive.google.com/file/d/12BDNBJQoU1N1GMqRq2yjJENaakRY0C_V/view?usp=sharing

ismailarilik, 2021-11-07:
It is an exe file and it is obvious that it is not a wiki so it may be a virus. If you want to look at the wiki you can navigate to https://github.com/ANYCUBIC-3D/Vyper/wiki If you want to download it you should [install git](https://git-scm.com/) and clone the wiki via the command `git clone https://github.com/ANYCUBIC-3D/Vyper.wiki.git`.

ismailarilik, 2021-11-07:
After clone you should change this line:

```md
![](images/board ac_tri_f1.png)
```

with this:

```md
![](images/board%20ac_tri_f1.png)
```

to be able to see the image.