ANYCUBIC-3D/Vyper #6: Firmware 2.3.9: when I click on "Home All" button of the "Move Axis" page, "Home All" button flickers for the full tome that the nozzle moves back to home

sanangeles84, 2021-08-03:
As per title, on firmware 2.3.9, when I click on "Home All" button of the "Move Axis" page, "Home All" button flickers for the full tome that the nozzle moves back to home.
It didn't happen with all the previous firmwares that I tried, by the way could be worthy to mention maybe that I didn't try to update the screen firmware

sergekatzmann, 2021-08-07:
@sanangeles84 
I have exactly the same problem. No update for screen either.

sanangeles84, 2021-08-07:
@sergekatzmann Actually I found other strange "animations" also in other situations, each time there is a long process where a button or a message stays on the screen for a while, I printed a couple of times and this looks to be mostly an aesthetical problem

sergekatzmann, 2021-08-08:
I don't use the screen that much, since I am controlling the printer with octoprint. the only thing I saw regarding message that was long time on was the confirmation dialog for the automatic bed leveling that was shown for a while during the leveling process. That was not the case before.
With this firmware I could print as well, but unfortunately I have alot of problems with my Vyper - mainboard fan was loud now it is not working anymore, inconsistent cooling of filament (have to tweak all the time), cracks found on z-motor mounts, so I couldn't probably tell if it is coming from firmware chnage or not.

sergekatzmann, 2021-08-08:
Just updated the screen software the problems seem to be solved now. I will observe if it is really the case. Printing benchy with the new firmware as a test now, looking alright so far.

sanangeles84, 2021-08-08:
@sergekatzmann I never done it before, I must remove plastic shell of the screen to access its SD card slot to update its firmware, right?

sergekatzmann, 2021-08-08:
You don't need to remove the shell. On the bottom side there is a card slot. Just look carefully to see what is the right side of the micro sd card to go up / down. Once you found the right position place it in the slot carefully and you will feel spring pressure on the card. Use something thin to help you to gentle push the card further into the slot till it locks.

jvonehr, 2021-10-03:
Note: the "right side" means contacts facing up, and to the back as you insert it. Tricky to get it in & out - I needed a pusher tool to get it to click out. There's a YouTube video on it: https://www.youtube.com/watch?v=NNJKI80F0i8