ANYCUBIC-3D/Vyper #7: Firmware v2.4.5 Bed Levelling

3DWolfdrive, 2021-09-20:
I have just tried upgrading to v2.4.5 with erasing EEPROM (0.0.6 motherboard), however I get an issue when trying to use automatic bed levelling function. The button on LCD starts flashing and it freezes, requiring a reboot. I have since moved back to v2.3.5 and bed levelling works there.

KAPTOHKA, 2021-09-20:
You have flashed the board with a new firmware, the screen also needs to be flashed with a new firmware and everything will work ...
We are all waiting for ANYCUBIC-3D to give us the source code of the screen and a tutorial for its cmpilation ...
PS. The difference between 2.3.5 and 2.4.5 versions on the board is only three digits, these are the coordinates of the parking of the nozzle and that's it ....

3DWolfdrive, 2021-09-20:
Thanks for the response. Everything else worked with the screen, so I thought it was an issue with the updated firmware. Hope we get the tutorial soon!

stuewe, 2021-09-23:
> You have flashed the board with a new firmware, the screen also needs to be flashed with a new firmware and everything will work, probably ...
> We are all waiting for ANYCUBIC-3D to give us the source code of the screen and a tutorial for its cmpilation ...
> PS. The difference between 2.3.5 and 2.4.5 versions on the board is only three digits, these are the coordinates of the parking of the nozzle and that's it ....

Anycubic released the 2.4.5 LCD firmware along with documentation today.

Here is a link in case someone can't find it: https://github.com/ANYCUBIC-3D/Vyper-lcd

KAPTOHKA, 2021-09-23:
> Anycubic released the 2.4.5 LCD firmware along with documentation today.

Yes ....? .... But there is nothing from ANYCUBIC, there is only DGUS-Tool of the ancient version and two usual examples... 
All this can be downloaded from the site DWIN-DGUS...

Cloud-121, 2021-09-24:
Just updated with new lcd and is still not working

elsbrock, 2021-09-25:
Looks like it's still a bit of chaos at Anycubic, hope they get it together soon.

Until this is fixed, you should be able to trigger bed leveling via `G29`, then store the mesh using `M500`.

see https://marlinfw.org/docs/gcode/G029-abl-bilinear.html

jvonehr, 2021-10-03:
I had the same issue - wish I had read this thread... How did you downgrade back to 2.3.5?

3DWolfdrive, 2021-10-03:
Luckily I was able to recover the old firmware file, so I just re-flashed the old one following the same steps. I haven't touched the LCD firmware as of yet. Hope we get an official fix or unified tools soon!

1800Power, 2021-11-07:
Hi, I´ve got the same problem.

How to solve this:

Use the Micro-SD-Card an fomat as in the README told.
Use FAT32 and 4096 Bytes!!!
Then copy the DWN_SET to the SD-Card.
Without Power put the SD-Card Chip-Side up in the bottom of the Display.
Now power up.
The update-procedure will take about 2 min (on my machine)

Kind reagards

Frank

bosco312, 2021-11-10:
I want to second the answer 1800Power gave, but add this:

It is important not to exceed the printers memory card capabilities. If you have a larger micro-SD, you should create a 4096 MB partition on it, formatted with 4096 bytes.

If you use too big a partition or do not format it correctly, the firmware update will still appear to work correctly. You will boot to the blue screen, the files will all show they downloaded, etc. However, there will be no beeping at the end, which you won't notice if you're not paying attention. Restarting the printer at this point will boot to the normal menu, but certain things won't work. Bed leveling being one of them. Trying to access that feature results in a flickering button and a locked up screen. Updating the firmware from this state, with the limitations noted above, will result in a correctly flashed and functioning machine. 

Side note: After all the updates, if you start and stop a print, then level the bed, you'll want to check the temperature settings first, and if the extruder temp is still 'on', select the 'cooling' button to turn it off. Otherwise you end up with a grid of sixteen very tiny nubs of drool on your build plate.

mixar, 2021-11-13:
> 
> 
> I had the same issue - wish I had read this thread... How did you downgrade back to 2.3.5?
> 
> Never mind - found the old stuff on their site: https://www.anycubic.com/blogs/videos/all-you-need-to-know-about-vyper

The files don't appear to be on their site any longer. Does anyone have a copy of 2.3.5?

Thanks

mixar, 2021-11-13:
Bosco312, Thank you !! This is the part I was missing.

> It is important not to exceed the printers memory card capabilities. If you have a larger micro-SD, you should create a 4096 MB partition on it, formatted with 4096 bytes.

oscar344765548, 2021-12-22:
Hi, I have the same problem and I don't know how to solve it, someone can help me since I'm new to 3d printing and I don't know anything about it.

Cloud-121, 2021-12-22:
If you still need help call me on discord. Cloud#5902

oscar344765548, 2021-12-22:
> 

ok I will write to you right now

chearner, 2022-01-03:
Is there a fix for the new firmware yet? 
I've tried to update screen firmware with no success, using various byte size formatted SD cards (FAT32).
I still see flashing bed leveling button.
I would downgrade my firmware but cannot find the old firmware on your site.
I found the github for 2.3.5, but do not see a .bin file.
Please display current and older firmware versions, always.

bosco312, 2022-01-03:
The cards you're formatting... Are they larger than 8 GB? If so, you'll
need to create and format a smaller partition on the SD card. I used a 4 GB
partition. Make sure the cluster size is not larger than 4096 bytes . That
should allow the Vyper to  correctly process the firmware update.

John Sheppard
206.660.1669
***@***.***

chearner, 2022-01-03:
I trying to format one now, I have 32GB and larger and Windows does not give an option for 4096 allocation size for some reason on any of these. 16k is lowest I've seen for FAT32. Trying in now on command line to see if it will take. Did the printer ship with a microSD card, I didn't see one? would be good to include a microSD card if this printer is going to be particular in this way.
Reason for upgrading firmware is due to printer stopping mid print, black screen, un responsive. Was printing overnight for large print. Happened 2 days in row now. I'm seeing this issue mentioned on Reddit forum recently. Printing remote via Octoprint. Could this be related to octoprint, would printing from SD card solve this? It is the large articulated dragon so it will be a 12+ hour print, large file. Printer stops dead, screen goes black.

ReverendRhyme, 2022-01-03:
As @bosco312  mentioned. You'll need an 8GB or lower card. If you have a larger card... You should be able to create an 8GB partition on the card. You can use the following video as a reference for Disk Management. https://youtu.be/Vtfo1PSYm_U

You'd want to create a partition of 8192 MB, and format that partition to 4096 bytes.

chearner, 2022-01-03:
Was able to get screen firmware to update after using partition tool to force 4k allocation. All good thanks. Going to try another print 12+ hours to see if printer stops again. Printer stops halfway thru last 2 prints and screen goes black, printer completely frozen. Going to try printing from SD card directly, not via octoprint. Thought it could be related to octoprint or power flicker at home during the night. Others are reporting this issue on Reddit, so it could be something bigger.

bosco312, 2022-01-03:
I have had no trouble printing via Octoprint, but you should have no trouble with using the SD card either. The only thing I'd add is that I've not tried to do a 12-hour print with the Vyper. I wish you much success.

Waffull, 2022-01-04:
I've switched to the community edition 6.1b firmware for Vyper. (
https://www.youtube.com/watch?v=fUq4NNnPE98)

Github: https://github.com/CR6Community/Marlin/tree/feat/vyper-support

Firmware files and instructions:
https://drive.google.com/file/d/1zrIIM6tdscbZGh9GdlL_jRJyzULWPxlC/view

I've successfully run 2 almost 24 hour prints with no problems.  ABL works,
Linear Advance works...  It seems quite stable.

I settled on the CE firmware after trying 2.4.5 as well as switching to
Klipper.  The lack of a functioning screen was a deal breaker for me with
klipper and I couldn't get klipperscreen to function for some reason.
Unfortunately there isn't a lot of info about CE for Vyper, at least in
English.  But there is a link to an English facebook group in the
description of the youtube video.  I don't use facebook and in fact it's
completely blocked on my network for security reasons, so I didn't check it
it, but there may be more info there.

Goodluck.

KAPTOHKA, 2022-01-04:
community edition 6.1b firmware for Vyper - пока что сыровата: 
- регулировка обдува только 0% или 100%
- не сохраняет настройки скоростей и рывков
- глюки с Octoprint
- длинные имена файлов не показывает
- кодировка шрифта в именах файлов только латиница
- Linear Advance на Bowden и тем более на вулкане работать не будет
- итд
В общем пока это только клон и разработчик просил много не ожидать...
Обещал в январе выкатить релиз...
Клонированием прошивки занимаются два с половиной человека и объединять усилия не хотят...

Arizona3D, 2022-05-18:
This post is somewhat related to the issues already posted here but different in that I have an older Firmware on my Vyper, and I experienced a strange issue that caused my auto bed leveling to not work and any subsequent prints to end in failures with the hot end gouging into the bed.  I was able to determine partially the cause of this problem with the help of 2 engineering buddies of mine.  A gcode file that I had sliced in Cura 4.10.0 on a Mac running OS 10.13.6 corrupts the EEPROM every time I try to print that specific file.  It somehow either overwrites the auto level net information or corrupts it so that the Vyper thinks that the bed is much lower than it actually is resulting in the nozzle being pressed into the bed by both Z motors which extreme force.  The fix is a reset of the EEPROM to factory defaults.  Just wanted to put this out there in case anyone else is experiencing this same problem.  It has taken me 5 months of off and on troubleshooting to find the cause and fix the issue.

fenua22, 2023-10-09:
i just upgrade my vyper to 2.4.5, i've the same problem with bed leveling.
Where can i find the old 2.3.5 firmware?

djdanby, 2024-10-11:
Same problem here. After update the Auto-leveling Button flickers when i press them. Card is formatted with Fat32 Standardsize (i think that is the correct one. There are no infos about the kb size). I use a 32GB SanDisk ultra. 
First, i updated the Mainboard and after that, the Display. The Problem came with Mainboard Update. Not with Display Update. :-/