ANYCUBIC-3D/Vyper #8: 2.3.5 & 2.4.5 Auto Bed Leveling doesn't work

stuewe, 2021-09-22:
Auto Bed Leveling doesn't seem to work on both 2.3.5 and 2.4.5. It probes fine until the 8th probe point, then a beeping sound is made and the printer homes. At that point the screen doesn't work anymore and it needs to be restarted.

This feature works fine on the official firmware, but on this Marlin firmware it does not.
The printer is stock, haven't changed anything on it.

The only change I did to the config was:
```C++
#define LIN_ADVANCE
#define LIN_ADVANCE_K 0
```

Any idea why this is the case? Does it work for anyone else?  

Best regards

stuewe, 2021-09-23:
Updating the LCD Firmware to 2.4.5 seems to have fixed the issue.

closed 2021-09-23 by stuewe

Toddi3D, 2022-05-05:
With Stock FW Linear Advance dont works. So you have to disable it.