ANYCUBIC-3D/Vyper #15: I want to add WiFi to the printer？

chinamummy, 2022-01-13:
I can now  WiFi control the printer.But The printer cannot read the TF card information on the module. I want to know whether the printer SD port is SPI or SDIO？
[image]
[image]
[image]
[image]

Hex994, 2022-04-11:
我也想更改增加wifi功能，具体要怎么操作？可以给联系方式请教一下吗？

chinamummy, 2022-04-11:
https://buyertrade.taobao.com/trade/detail/tradeSnap.htm?spm=a1z09.2.0.0.67002e8dyrmxXk&tradeID=2408747940626004431&snapShot=true可以去这家店铺了解一下

Hex994, 2022-04-13:
> [https://buyertrade.taobao.com/trade/detail/tradeSnap.htm?spm=a1z09.2.0.0.67002e8dyrmxXk&tradeID=2408747940626004431&snapShot=true可以去这家店铺了解一下](https://buyertrade.taobao.com/trade/detail/tradeSnap.htm?spm=a1z09.2.0.0.67002e8dyrmxXk&tradeID=2408747940626004431&snapShot=true%E5%8F%AF%E4%BB%A5%E5%8E%BB%E8%BF%99%E5%AE%B6%E5%BA%97%E9%93%BA%E4%BA%86%E8%A7%A3%E4%B8%80%E4%B8%8B) 在2022-04-11 15:29:59，” Hex9994" ***@***.***> 英文是：我也想添加wifi，具体怎么做？或取消订阅。您收到此消息是因为您要编写该功能。 ID： ***@***.***>

显示:"对不起，你无权查看该宝贝的交易信息"

alienwaredream, 2022-04-25:
我自己做了一个WIFI SD卡 类似这样的https://github.com/FYSETC/ESPWebDAV
插在VYPER上也是无法识别到卡内容
这个项目用的SPI模式，所以....应该和你是类似的问题？