ANYCUBIC-3D/Vyper #25: Inverse Extruder Motor, can't compile

whatwecee, 2022-05-07:
I am attempting to inverse the extruder motor but cannot compile to create a stable build.

The only detail that changed is that I adjusted the extruder line 1130 of the configuration.h to be false instead of true. This is because I have a new extruder that requires the opposite directionality.

[image]
[image]

What did I do wrong? what would I need to adjust? printer is default settings for everything besides this one thing I'm trying to change

whatwecee, 2022-05-07:
To followup, if I go back and change the 1130 line of the configuration.h file to true, I still get those errors

Toddi3D, 2022-05-11:
You have to change the pins in the File for it.

Toddi3D, 2022-05-11:
You have a lot of errors, but you can read it where the errors are located.